awtrix/AWTRIX2.0-Controller #3: PCB available?

BlueAndi, 2019-05-10:
Will there be a PCB provided, containing all the sensors and etc.?

Blueforcer, 2019-05-19:
Yes, its in development right now
[image]

BlueAndi, 2019-05-20:
Looks great. Can I see the schematic anywhere?
Is there a fuse and a supply reverse connect protection on it?
And the important question, when do you expect to release it? ;-)

Mazze2000, 2019-06-13:
> Is there a fuse and a supply reverse connect protection on it?
Yes, it is ;-)

BlueAndi, 2019-06-14:
Thanks Matthias for the reply. In the meantime I started with a PCB too, but based on an ESP32.

closed 2019-06-14 by BlueAndi