awtrix/AWTRIX2.0-Controller #4: Cannot build on esp8266 Arduino

LockonS, 2019-05-17:
FastLED fail to compile with following errors:

```
Processing nodemcuv2 (framework: arduino; platform: espressif8266; board: d1_mini)
---------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI> 1.0
|-- <ArduinoOTA>
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|-- <Wire> 1.0
|-- <FastLED> 3.2.6
|   |-- <EspSoftwareSerial> 5.0.4
|-- <ESP8266WiFi> 1.0
|-- <LDR>
|-- <PubSubClient> 2.7
|-- <DFPlayerMini_Fast>
|-- <EspSoftwareSerial> 5.0.4
|-- <FastLED NeoMatrix> 1.0
|   |-- <Adafruit GFX Library> 1.2.3
|   |   |-- <SPI> 1.0
|   |-- <FastLED> 3.2.6
|   |   |-- <EspSoftwareSerial> 5.0.4
|-- <ArduinoJson> 5.13.4
|-- <SparkFun APDS9960 RGB and Gesture Sensor> 1.4.2
|   |-- <Wire> 1.0
|-- <Adafruit GFX Library> 1.2.3
|   |-- <SPI> 1.0
Compiling .pioenvs/nodemcuv2/lib27a/FastLED/FastLED.cpp.o
Compiling .pioenvs/nodemcuv2/lib27a/FastLED/bitswap.cpp.o
Compiling .pioenvs/nodemcuv2/lib27a/FastLED/colorpalettes.cpp.o
Compiling .pioenvs/nodemcuv2/lib27a/FastLED/colorutils.cpp.o
Compiling .pioenvs/nodemcuv2/lib27a/FastLED/hsv2rgb.cpp.o
Compiling .pioenvs/nodemcuv2/lib27a/FastLED/lib8tion.cpp.o
Compiling .pioenvs/nodemcuv2/lib27a/FastLED/noise.cpp.o
Compiling .pioenvs/nodemcuv2/lib27a/FastLED/power_mgt.cpp.o
In file included from lib/FastLED/led_sysdefs.h:43:0,
from lib/FastLED/FastLED.h:34,
from lib/FastLED/FastLED.cpp:2:
/Users/LockonStratos/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:191:14: error: conflicting declaration 'typedef bool boolean'
typedef bool boolean;
^
In file included from lib/FastLED/led_sysdefs.h:27:0,
from lib/FastLED/FastLED.h:34,
from lib/FastLED/FastLED.cpp:2:
lib/FastLED/platforms/esp/8266/led_sysdefs_esp8266.h:15:17: error: 'boolean' has a previous declaration as 'typedef uint8_t boolean'
typedef uint8_t boolean;
^
In file included from lib/FastLED/led_sysdefs.h:43:0,
from lib/FastLED/FastLED.h:34,
from lib/FastLED/bitswap.cpp:2:
/Users/LockonStratos/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:191:14: error: conflicting declaration 'typedef bool boolean'
typedef bool boolean;
^
In file included from lib/FastLED/led_sysdefs.h:27:0,
from lib/FastLED/FastLED.h:34,
from lib/FastLED/bitswap.cpp:2:
lib/FastLED/platforms/esp/8266/led_sysdefs_esp8266.h:15:17: error: 'boolean' has a previous declaration as 'typedef uint8_t boolean'
typedef uint8_t boolean;
^
In file included from lib/FastLED/led_sysdefs.h:43:0,
from lib/FastLED/FastLED.h:34,
from lib/FastLED/colorpalettes.cpp:4:
/Users/LockonStratos/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:191:14: error: conflicting declaration 'typedef bool boolean'
typedef bool boolean;
^
In file included from lib/FastLED/led_sysdefs.h:27:0,
from lib/FastLED/FastLED.h:34,
from lib/FastLED/colorpalettes.cpp:4:
lib/FastLED/platforms/esp/8266/led_sysdefs_esp8266.h:15:17: error: 'boolean' has a previous declaration as 'typedef uint8_t boolean'
typedef uint8_t boolean;
^
In file included from lib/FastLED/led_sysdefs.h:43:0,
from lib/FastLED/FastLED.h:34,
from lib/FastLED/colorutils.cpp:7:
/Users/LockonStratos/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:191:14: error: conflicting declaration 'typedef bool boolean'
typedef bool boolean;
^
In file included from lib/FastLED/led_sysdefs.h:27:0,
from lib/FastLED/FastLED.h:34,
from lib/FastLED/colorutils.cpp:7:
lib/FastLED/platforms/esp/8266/led_sysdefs_esp8266.h:15:17: error: 'boolean' has a previous declaration as 'typedef uint8_t boolean'
typedef uint8_t boolean;
^
In file included from lib/FastLED/led_sysdefs.h:43:0,
from lib/FastLED/FastLED.h:34,
from lib/FastLED/hsv2rgb.cpp:4:
/Users/LockonStratos/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:191:14: error: conflicting declaration 'typedef bool boolean'
typedef bool boolean;
^
In file included from lib/FastLED/led_sysdefs.h:27:0,
from lib/FastLED/FastLED.h:34,
from lib/FastLED/hsv2rgb.cpp:4:
lib/FastLED/platforms/esp/8266/led_sysdefs_esp8266.h:15:17: error: 'boolean' has a previous declaration as 'typedef uint8_t boolean'
typedef uint8_t boolean;
^
In file included from lib/FastLED/led_sysdefs.h:43:0,
from lib/FastLED/FastLED.h:34,
from lib/FastLED/lib8tion.cpp:3:
/Users/LockonStratos/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:191:14: error: conflicting declaration 'typedef bool boolean'
typedef bool boolean;
^
In file included from lib/FastLED/led_sysdefs.h:27:0,
from lib/FastLED/FastLED.h:34,
from lib/FastLED/lib8tion.cpp:3:
lib/FastLED/platforms/esp/8266/led_sysdefs_esp8266.h:15:17: error: 'boolean' has a previous declaration as 'typedef uint8_t boolean'
typedef uint8_t boolean;
^
In file included from lib/FastLED/led_sysdefs.h:43:0,
from lib/FastLED/FastLED.h:34,
from lib/FastLED/noise.cpp:2:
/Users/LockonStratos/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:191:14: error: conflicting declaration 'typedef bool boolean'
typedef bool boolean;
^
In file included from lib/FastLED/led_sysdefs.h:27:0,
from lib/FastLED/FastLED.h:34,
from lib/FastLED/noise.cpp:2:
lib/FastLED/platforms/esp/8266/led_sysdefs_esp8266.h:15:17: error: 'boolean' has a previous declaration as 'typedef uint8_t boolean'
typedef uint8_t boolean;
^
In file included from lib/FastLED/led_sysdefs.h:43:0,
from lib/FastLED/FastLED.h:34,
from lib/FastLED/power_mgt.cpp:2:
/Users/LockonStratos/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:191:14: error: conflicting declaration 'typedef bool boolean'
typedef bool boolean;
^
In file included from lib/FastLED/led_sysdefs.h:27:0,
from lib/FastLED/FastLED.h:34,
from lib/FastLED/power_mgt.cpp:2:
lib/FastLED/platforms/esp/8266/led_sysdefs_esp8266.h:15:17: error: 'boolean' has a previous declaration as 'typedef uint8_t boolean'
typedef uint8_t boolean;
^
*** [.pioenvs/nodemcuv2/lib27a/FastLED/FastLED.cpp.o] Error 1
*** [.pioenvs/nodemcuv2/lib27a/FastLED/bitswap.cpp.o] Error 1
*** [.pioenvs/nodemcuv2/lib27a/FastLED/colorpalettes.cpp.o] Error 1
*** [.pioenvs/nodemcuv2/lib27a/FastLED/lib8tion.cpp.o] Error 1
*** [.pioenvs/nodemcuv2/lib27a/FastLED/hsv2rgb.cpp.o] Error 1
*** [.pioenvs/nodemcuv2/lib27a/FastLED/colorutils.cpp.o] Error 1
*** [.pioenvs/nodemcuv2/lib27a/FastLED/power_mgt.cpp.o] Error 1
*** [.pioenvs/nodemcuv2/lib27a/FastLED/noise.cpp.o] Error 1
================================================ [ERROR] Took 2.40 seconds ================================================
The terminal process terminated with exit code: 1
```
And a workaround was posted here by @CRCinAU [link](https://github.com/FastLED/FastLED/issues/733), commenting out a single line makes everything work.

File: path-to-project/platforms/esp/8266/led_sysdefs_esp8266.h
Line: 15
Change to: //typedef uint8_t boolean;

Blueforcer, 2019-05-19:
There will be an update the next days, to work with the latest ESP8266 Core

LockonS, 2019-05-26:
Thanks, all the problems was solved with this latest update.

closed 2019-06-12 by Blueforcer