awtrix/AWTRIX2.0-Controller #8: No Matrix on port /dev/ttyUSB0

zlinwei, 2019-07-17:
No Matrix on port /dev/ttyUSB0

I edit the config:
```cpp
const char *ssid = "xxx";
const char *password = "xxx";
char *awtrix_server = "xxx";

#define USB_CONNECTION
// #define MATRIX_MODEV2

/// LDR Config
#define LDR_RESISTOR 1000 //ohms
#define LDR_PIN A0
#define LDR_PHOTOCELL LightDependentResistor::GL5516
```

on server i set usb port equals /dev/ttyUSB0  
but i got an error;
Thanks;

Mazze2000, 2019-07-17:
Do you try the autoconnect?
Please show me the log of the Server.

zlinwei, 2019-07-17:
when I set USBPort=auto
the logs:
```
11:39:33 PM      AWTRIX 2.0 by Stephan Muehl
11:39:33 PM      01619
No Matrix on port /dev/ttyAMA0
open port /dev/ttyUSB0
port /dev/ttyUSB0 opened
waiting for answer
No Matrix on port /dev/ttyUSB0
```

Mazze2000, 2019-07-17:
I don‘t really know where the Problem is.
Maybe you try it with wifi First.

Do you Connect the Controller via USB cable or direct with Rx Tx and GND?

zlinwei, 2019-07-18:
I connect the controller via USB cable; Can I use Serial to connect the ESP8266 and my raspberry pi?

Mazze2000, 2019-07-18:
Yes you can use one of it.
Do you try the connectrion via wifi?
Please note: If you change the connection in the server, you have to restart the server.

zlinwei, 2019-07-20:
Yes, It works when I use WIFI, but USB didn't, Thanks so much!

zlinwei, 2019-07-20:
emmm, it works via USB on my macos, 
[image]

closed 2020-04-28 by Blueforcer