awtrix/AWTRIX2.0-Controller #20: about the new version

CodeDZCpt, 2020-03-15:
1. what's the server login password? 
2. what's the reason that my LED is lighting `Host ->` since I flash the .bin

thank you

CodeDZCpt, 2020-03-15:
[image]

CodeDZCpt, 2020-03-15:
[IMG_4534.pdf](https://github.com/awtrix/AWTRIX2.0-Controller/files/4333880/IMG_4534.pdf)

Blueforcer, 2020-03-17:
Because it searching for the host:)

CodeDZCpt, 2020-03-18:
thanks， Can't it connect the server?

stgapr, 2020-03-21:
The LED light on ESP8266 blinks continuously when connected to power, and there's a SSID called AI_THINKER-D89824. 
As connected, http://172.17.114.17:7000  shows login interface absolutely the same as above when the SSID connected. What should I do?

Blueforcer, 2020-03-21:
> The LED light on ESP8266 blinks continuously when connected to power, and there's a SSID called AI_THINKER-D89824.
> http://172.17.114.17:7000 shows login interface absolutely the same as above when the SSID connected. What should I do?
> 
> ps: It seems I didn't flash firmware right. The SSID "AI_THINKER-xxx" is default name of hardware producer NodeMCU.

Erase the ESP, and reflash it like described in the documentation
https://awtrixdocs.blueforcer.de/#/en-en/controller

Keep in mind, that the firmware.bin is created for an Wemos D1. On other ESP variants it might not work and you have to modify the firmware yourself (build platform in the platform.io config)

closed 2020-04-28 by Blueforcer