awtrix/AWTRIX2.0-Controller #32: Can you publish a source code that can be compiled with Arduino instead of vscode

lixy123, 2020-08-17:
I successfully compiled this application after installing the appropriate library with version 1.8.12 Arduino. Several libraries of neomatrix can not use the latest version, otherwise they can't be compiled. I have been struggling for a long time. Strangely, the LED light is not on. Can you upgrade the library files related to led in the next development. But I burn it https://blueforcer.de/awtrix/stable/firmware.bin The firmware works properly. I even suspect that the compiled firmware you released is not this code

Blueforcer, 2020-11-26:
There is no sourcecode for Arduino IDE. Its written in VSC with PlatformIO. So if you want to chnage the code, you have to use VSC

closed 2020-11-26 by Blueforcer

lixy123, 2020-11-27:
After some efforts, awtrix can be successfully compiled on Arduino IDE, which requires some code adjustment. There are some problems with the touch function, and the overall operation is very satisfactory. This software is my favorite system in raspberry pie besides camera monitoring program and homeassistant