awtrix/AWTRIX2.0-Controller #33: Frequently show error log during awtrix controller operation

GeorgeLeungHK, 2020-08-20:
The controller version was updated to 0.35, and used x86 PC for awtrix’s host in Docker "whyet-awtrix21" (Operation System: Synology DSM 6.1.7). but error logs in the host were shown repeatedly. And the host would be hanged up about 1 hour.

Error logs were listed as follow:-

java.awt.AWTException: headless environment
	at java.awt.Robot.<init>(Robot.java:91)
	at butt.droid.awtRobot.AWTRobot.<init>(AWTRobot.java:87)
	at de.awtrix.webstats._class_globals(webstats.java:123)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at anywheresoftware.b4a.BA.raiseEvent2(BA.java:91)
	at de.awtrix.webstats.innerInitialize(webstats.java:22)
	at de.awtrix.webstats._initialize(webstats.java:195)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at anywheresoftware.b4j.object.JServlet.createInstance(JServlet.java:62)
	at anywheresoftware.b4j.object.WebSocketModule$Adapter$ThreadHandler.run(WebSocketModule.java:178)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)

GeorgeLeungHK, 2020-08-20:
Moreover, if all apps in general setting were setup "start time" and "end time", repeatedly errors would be shown when the actual time was out of the setup time.

Error logs were listed as below. 

	at de.awtrix.main$ResumableSub_ChangeApp.resume(main.java:1022)
	at de.awtrix.main._changeapp(main.java:332)

Blueforcer, 2020-11-26:
this has nothing to do with the controller, please open a post in the forum

closed 2020-11-26 by Blueforcer