awtrix/AWTRIX2.0-Controller #37: AWTRIX HOST IP SET FAILED

AICrazy, 2021-04-25:
Hello,

Every time I configure wifi and HOST IP in this step, wifi ap can be set successfully, but HOST IP configuration fails, HOST IP is always the default value 0.0.0.0.

Thank you.

[image]

BornTKill, 2021-08-07:
@AICrazy Add same issue after upgrading. To fix I erase the esp and flash again with firmware download on documentation page. Good Luck

jh163888, 2021-08-21:
Same issue here, but I have no access to USB-COM port any more since I 3D print the whole clock and left no debug port.. lol Waiting for the fix to upgrade through wifi.

BattloXX, 2021-12-08:
I have the sam Problem with the update

Blueforcer, 2022-03-26:
Yep, in some cases there are old files available from older flashed Programms. 
Erase the complete esp and you will be fine

closed 2022-03-26 by Blueforcer