awtrix/AWTRIX2.0-Controller #45: Boot loop after uploading a compiled binary

bench2012, 2021-09-21:
Hi,

I am not sure if I am the only one here, I compile from source using platform.io (after taking out the include String.h in Menucontrol.cpp, otherwise it will not compile). Uploaded firmware.bin to an nodemcu v1, (tried using both platformio and cli) and now it is in a boot loop with:

rst cause:2, boot mode:(3,7)

Any idea?

Bench

avanx, 2021-09-23:
See(https://github.com/awtrix/AWTRIX2.0-Controller/issues/44), add switch when size equals zero 
Besides,
try to change configuration in **platformio.ini**
from
`platform = espressif8266`
to
`platform = espressif8266@2.6.3`
I guess pre-compile binary was compiled using 8266Arduino SDK 2.x.Doing these helps me to stop rebooting and fixes issure (https://github.com/awtrix/AWTRIX2.0-Controller/issues/43)

bench2012, 2021-09-29:
Thx, will give this a try.

bench2012, 2021-09-29:
Hi,

Try it and added a quick "fix" to setup:

if (size !=0)
			{
			JsonObject &json = jsonBuffer.parseObject(buf.get());
			...............
			}
			configFile.close();

Not sure if this damage anything, but it compiled ok and I change te SDK to 2.6.3 and it is now working.

Thanks for the support and a great product.

closed 2021-09-29 by bench2012