awtrix/AWTRIX2.0-Controller #57: Impossible to connect controler - saving wifi settings stuck

benlec, 2022-05-06:
Hello,
I have tried multiple times to connect my controller.
I enter all the wifi information router, password, host ip and save.
It stays to the next screen where is says saving wifi and that's all....
it never really connects.
I reflashed the ESP, restarted the whole process but it does not work.
my host is hosted on internet but accessible from my home network.
Is this an issue?
Thanks

benlec, 2022-05-11:
Anyone has any idea about this? Any troubleshooting steps we can try?

RalfJahns, 2022-06-13:
The ESP board gives a lot of information on the serial port. So I would recommend to connect a PC by USB cable and use putty or so (I think it's 115200 baud) to inspect this.