awtrix/AWTRIX2.0-Controller #60: Brightness control doesn't work anymore

no-Legend, 2022-10-09:
Since the last updates, it looks like that the brightness control at my awtrix classic isn't working anymore.

Also if the project is closed, it would be great to get a bug fix for that problem.

Cheers Robert

cuijiemmx, 2022-11-02:
@no-Legend Update both the firmware and host to the latest version will resolve this issue

HollyGhos, 2022-11-03:
@cuijiemmx - Updating my Matrix to 0.46 and Host to 2045 causes this issue with flickering brightness on my side!
May I ask you what versions you are using?
Thx

RalfJahns, 2022-12-02:
Same problem here. When app is switched from one to the other, brightness goes to 100%. After a few seconds, it switches back to "normal" brightness (I have LDR installed). Bugfix would be fine ...